zivillian/esp32-modbus-gateway #2: Disconnect Timeout

agisofttm, 2022-06-24:
Hi @zivillian 

Really nice, simple and working gateway but there is some little problems. First one is when client is connected to the gateway, after 10 seconds is disconnected as the debug says: 
[11:28:38][D] 93697| ModbusServerTCPtemp.h [ 239] serve: Accepted connection - 1 clients running
[11:28:48][D] 103686| ModbusServerTCPtemp.h [ 350] worker: Worker stopping due to timeout.

Second one, can you integrate hardware gpio button for getting config portal on demand? Because with Wifi Manager when you set once SSID, then you cannot make a config to anything now.

Thank you
Kind Regards

zivillian, 2022-06-24:
Hi @agisofttm,

thanks for your feedback. I've already added a wifi reset button to the home page, but I'm not sure if this suits your needs. If the esp can't conntect to the saved wifi it will also start the config portal.

Regarding the timeout. This is only an inactivity timeout. As long as there is modbus traffic goint on, the connection is kept alive, but after 10s of silence the connection is dropped to cleanup stale resources. Would it help to make this timeout configurable?

Regards
zivillian

agisofttm, 2022-06-25:
Hi @zivillian,

Yes Wifi Reset on web page is do the thing, but because home page is missing after first configuration there is no way to reset settings or to debug. For that reason I added this on loop 

`void loop() {
  // put your main code here, to run repeatedly:
  webServer.begin();`

and now I have access to home page via new IP on already configured SSID.

Regarding Timeout - it will be amazing if you add this timeout configurable, because a lot of devices that use modbus are tested slowly for unknown registers (for example HVAC).

Also I found that you use function for any function modbus reading 
`MBbridge.attachServer(i, i, ANY_FUNCTION_CODE, &MBclient);`
which is great, but on debug section there are only Functions 03 and 04 which reads registers only. Is it possible to add different functions to debug as well? Like Read Coils, Write Multiple Coils, Write Multiple Registers and etc... 
I'm trying to communicate with a lot of HVAC, and depend on brand they have different needs.

Thank you very much :-)

zivillian, 2022-06-26:
I've added a config option for the tcp timeout. Can you please test the build from [here](https://github.com/zivillian/esp32-modbus-gateway/actions/runs/2565204410)

zivillian, 2022-06-26:
I'll take a look into the problem, that the webui is not reachable after initial wifi configuration.

Regarding the debug page - initially i wanted to suppoert all functions, but started with those two. For yome Functions additional input is needed and didn't know how the web ui should look like, so this is only the first draft. If you have an idea how the input and output should be laid out, please open a separate issue or pr.

agisofttm, 2022-06-27:
@zivillian 
> I've added a config option for the tcp timeout. Can you please test the build from [here](https://github.com/zivillian/esp32-modbus-gateway/actions/runs/2565204410)

Interesting is that WEB server appears, after I installed firmware from your link, but unfortunately when I change Timeout to 20000 (20sec) it show to save config, but after testing is still 10seconds - here is log:

[11:30:30][webserver] GET /style.css
[11:31:03][webserver] GET /
[11:31:03][webserver] GET /style.css
[11:31:04][webserver] GET /config
[11:31:04][webserver] GET /style.css
[11:31:09][webserver] POST /config
[11:31:09][webserver] saved port
[11:31:09][webserver] saved timeout
[11:31:09][webserver] saved modbus baud rate
[11:31:09][webserver] saved modbus data bits
[11:31:09][webserver] saved modbus parity
[11:31:09][webserver] saved modbus stop bits
[11:31:09][webserver] saved serial baud rate
[11:31:09][webserver] saved serial data bits
[11:31:09][webserver] saved serial parity
[11:31:09][webserver] saved serial stop bits
[11:31:09][webserver] GET /
[11:31:09][webserver] GET /style.css
[11:31:18][D] 433857| ModbusServerTCPtemp.h [ 261] worker: Worker started, timeout=10000
[11:31:18][D] 433857| ModbusServerTCPtemp.h [ 210] accept: Started client 0 task 1073626432
[11:31:18][D] 433868| ModbusServerTCPtemp.h [ 239] serve: Accepted connection - 1 clients running
[11:31:28][D] 443857| ModbusServerTCPtemp.h [ 350] worker: Worker stopping due to timeout.

zivillian, 2022-06-27:
you need to reboot to apply the config changes

agisofttm, 2022-06-27:
> you need to reboot to apply the config changes

PErfect - thanks it is working now. I'll do further tests and give a feedback :-)))

zivillian, 2022-06-28:
Hey @maufc,

thanks for your feedback!

1. I've already added a button in WebUI to reset the wifi config. Do you need an additional hardware button for this? After a blackout the device will open the config portal, but the config is not deleted - just power cycle and it will reconnect to known SSID.
2. I'll take a look into disabling the upload button if no file is selected.
3. I've not tested this yet, because currently I have some hardware issues, but as soon as those are resolved I plan to query up to 320 register. If this is an issues, this has to be resolved - just give me some time.
4. I'm using the bridge functionality of [eModbus](https://github.com/eModbus/eModbus), which requires you to setup a mapping of slave ids from TCP to RTU. Instead of adding a complex config UI I chose to map all 257 slave ids 1:1. That's what you see in the serial output.
5. The error you get is [`ILLEGAL_DATA_ADDRESS`](https://github.com/eModbus/eModbus/blob/8a8da1f5abffcd91520e2446e1bb3bf4f6ab0fba/src/ModbusTypeDefs.h#L70). From the screenshot it  look's like you've selected the wrong function. Your instsruction state `04 Read input register` but you've selected `03 Read Holding Register`.
6. Ethernet is a great idea - I had the same a few day ago. We are currently evaluating the hardware options to add ethernet to an esp32. As soon as my prototype has as ethernet port, I will add support for ethernet to the firmware.
7. I don't think it's possible to support ESP8266, because all the Modbus RTU stuff is handled by eModbus which soes not support ESP8266 (because it has no additional HardwareSerial).

Once again, thanks for your feedback. It helps to focus on the relevant parts...

Regards
zivillian

agisofttm, 2022-06-29:
@maufc @zivillian I'm also think about ESP8266, but let's focus to run everything on ESP32 and after I'll try to port it to ESP8266 :-))) Also I found something interesting here. I use different types of MODBUS boards. One is cheap china MAX485 board without automatic flow control, and I need to create additional hardware to handle requests. Second one have a full TTL and no need to create anything. With first board there is an often error 0x2 but with second one there is no error, so hardware is important for eModbus I think.

maufc, 2022-07-04:
Hello @zivillian 

1.I would say yes, a hardware button will be useful on site, when something happened to the connection and you don't got a computer to connect the device and completely flash it. What choices do you have in this case. Am I missing some feature of WiFiManager?
You are right, after a blackout it will continue working as long as both devices are down. But if only the router is down for any reason?
2. Great!
3. So, at this moment, this is a limitation? the number of registers you can query. Isn't it a timeout issue?
4. Understood.
5. I tried both, with same result. My device accepts both, too. There is something more there. CRC?
6. Great!. The ethernet interface should be SPI, as popular chips as W5100 and W5500 works with it. I would add here, for both WiFi and Ethernet options, an additional feature to use DHCP and fixed IP.
7. The emodbus documentation says that async versions runs on ESP8266. Wouldn't be possible to work with software serial? Anyway, I agree with @agisofttm that would be better to focus on ESP32.
8.  @agisofttm, what you are using are not different modbus boards, but different RS-485 boards. I particularly use the auto flow control board with no problems. I am pretty sure that you need additional programming to manage the selection pin. 
Thanks to both!

maufc, 2022-07-04:
Hi again.
I have to ask one additional feature:
9. The web should be user/password protected.

maufc, 2022-07-05:
I took a look to eModbus documentation.
The eModbus library already has a function to work with selection pin boards @agisofttm .
Ethernet feature is also develop.
Just to comment...

zivillian, 2022-07-25:
I've added a config option for an (optional) RTS pin, but since all of my hardware does AutoRTS I'm not able to test it.

zivillian, 2022-07-25:
I can confirm, that reading more than 20 registers shows a significant error rate, but I fear, that this might be caused by eModbus/eModbus#198

zivillian, 2023-02-15:
the errors should be fixed with https://github.com/eModbus/eModbus/issues/268