zivillian/esp32-modbus-gateway #4: Support for M5Stack ATOM RS485

wlcrs, 2022-08-18:
The M5Stack ATOM RS485 has the RX/TX pins connected to GPIO 22/19. 

To get this project to work, I had to make the following modifications:

```patch
diff --git a/platformio.ini b/platformio.ini
index b8717e9..4c62481 100644
--- a/platformio.ini
+++ b/platformio.ini
@@ -10,7 +10,7 @@

 [env:esp32dev]
 platform = espressif32
-board = esp32dev
+board = m5stack-atom
 framework = arduino
 lib_deps =
     https://github.com/tzapu/WiFiManager.git
diff --git a/src/main.cpp b/src/main.cpp
index 86c15dc..edafbcc 100644
--- a/src/main.cpp
+++ b/src/main.cpp
@@ -30,7 +30,7 @@ void setup() {
   wm.autoConnect();
   dbgln("[wifi] finished");
   dbgln("[modbus] start");
-  modbusSerial.begin(config.getModbusBaudRate(), config.getModbusConfig());
+  modbusSerial.begin(config.getModbusBaudRate(), config.getModbusConfig(), GPIO_NUM_22, GPIO_NUM_19);
   MBclient = new ModbusClientRTU(modbusSerial, config.getModbusRtsPin());
   MBclient->setTimeout(1000);
   MBclient->begin();
```

Without these modifications, the firmware crashed during initialization:

```
[wifi] finished
[modbus] start
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57
```

Any chance that you can prevent this crashing from occurring and make the RX/TX pins configurable?

zivillian, 2022-08-21:
Thanks for reporting. Can you test the firmware from #5 (available [here](https://github.com/zivillian/esp32-modbus-gateway/actions/runs/2899847747))?

pantherale0, 2025-03-31:
@zivillian sorry for bringing up an old issue, m5 stack does not work for me, fresh build, with platformio changes made.

[logs.txt](https://github.com/user-attachments/files/19535818/logs.txt)