zivillian/esp32-modbus-gateway #18: ESp stalls after ? time

RalfWein, 2023-09-10:
Hi zivilian, hi everybody,
thank you for creating and sharing this nice gateway!

I tried to run esp32-modbus-gateway V0.0.4 on two ESP32 WROOM, which works nice, but after some time (seconds, hours or days) the ESP stalls. Am I'm the only one with this issue? Could it be a memory leak?
Tried several different hardware to exclude a damaged ESP causing the problem.
Would it be feasable to add the internal WDT?

Ralf

Fleckz, 2023-10-02:
You are not alone, same happening here, I don't even communicate trough RS485, just sits there connected to the WiFi and after a while, seems like it's getting slower and slower, drops some ping packets more and more, then gone at all, ideas what's causing this?

zivillian, 2023-10-02:
Can you please try again with the v0.0.5 release?

Fleckz, 2023-10-03:
I'm sorry, maybe I am dumb, but I could not find version number anywhere... maybe this helps somehow:
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- WiFiManager @ 2.0.16-rc.2+sha.6a92492
|-- ESP Async WebServer @ 1.2.3+sha.f71e3d4
|-- eModbus @ 1.7.0+sha.ed34322
|-- AsyncTCP @ 1.1.1
|-- Preferences @ 2.0.0
|-- WiFi @ 2.0.0
|-- Update @ 2.0.0

Maybe you can send .bin I coul try to flash?
Or guide me how to debug this maybe? Thanks anyway!

zivillian, 2023-10-03:
> Maybe you can send .bin I coul try to flash?

Please try the binary from the latest Release at https://github.com/zivillian/esp32-modbus-gateway/releases/tag/v0.0.5

Fleckz, 2023-10-03:
Thanks, now I have the following:
Build time:	Jul 31 2023 10:22:11
Will see

RalfWein, 2023-10-03:
Hi zivilian, hi everybody,

thank you!
I've installed V0.0.5 now. What was/is the issue?
But , as I wrote: the stall happened after minutes/hours/weeks ...
Let's see.

Ralf

zivillian, 2023-10-11:
@RalfWein I've moved all the modbus stuff to the second core. I'm not sure if this was the root cause, so please report back if you still have issues.

RalfWein, 2023-10-12:
@zivillian No stall for a week, much faster, less errors! I'm optimistic. Thank you very much!

closed 2023-10-12 by RalfWein