zivillian/esp32-modbus-gateway #24: current build not compiling

chilltrout, 2024-05-21:
Hi 
Just wondering if anyone else can get the current build to compile in platfromio 
seems like async might be out of date or something

zivillian, 2024-05-28:
the github actions are also no longer working: https://github.com/zivillian/esp32-modbus-gateway/actions/runs/9274726451

zivillian, 2024-05-28:
the failing github action is complaining about errors in WiFiManager

> error: format '%lu' expects argument of type 'long unsigned int', but argument 3 has type 'uint32_t' {aka 'unsigned int'}

which is caused by the `-Werror` flag and already addressed by the (not yet merged) PR https://github.com/tzapu/WiFiManager/pull/1740

zivillian, 2024-10-12:
fixed in 1ed404a

closed 2024-10-12 by zivillian

Cavelynx, 2025-03-13:
I am encountering this issue too. I have been trying the latest release [b51c58b](https://github.com/zivillian/esp32-modbus-gateway/commit/b51c58b96f085bb021c27c8838a1601b2956df2e) as well as [1ed404a](https://github.com/zivillian/esp32-modbus-gateway/commit/1ed404a641e41aae7be7843248205a88397cf21c) and am still unable to compile them, is there a workaround for this?

zivillian, 2025-03-16:
@Cavelynx can you share your error message?

Cavelynx, 2025-03-17:
> [@Cavelynx](https://github.com/Cavelynx) can you share your error message?

I'm getting the same errors as the OP. This was using the V0.0.6 release: 

[image]