zivillian/esp32-modbus-gateway #26: Culpability with max chips where flow control pin is required?

fgampel, 2024-06-15:
Hello, 
Thanks for your great work. 
Would it be possible to adapt this code somehow to make it compatible with max chips where we need the third pin for flow_control?

Thanks

zivillian, 2024-06-16:
The RTS pin can be set in the config UI
[image]

fgampel, 2024-06-16:
Hello, 
Thanks very much for your answer.
I'm receving the following error while trying to compile on platformio
[image]

Besides platformio says that all the dependencies are updated
[image]

Would you please point me which is the library you used for this pins_arduino.h so I can use exactly the same as you?

Thanks in advance

fgampel, 2024-06-16:
I'm getting now exactly the same compilation error as the github platform on https://github.com/zivillian/esp32-modbus-gateway/actions/runs/9275275741/job/25519578175

[image]

fgampel, 2024-06-16:
extra info, editing line 3368 on Wifimanager.cpp changing
_debugPort.printf("[MEM] free: %5lu | max: %5u | frag: %3u%% \n", free, max, frag);
to 
_debugPort.printf("[MEM] free: %5u | max: %5u | frag: %3u%% \n", free, max, frag);

fixed the compilation error:
[image]

closed 2025-01-07 by zivillian